AllskyTeam/allsky-modules #1: Error - Allsky not installed

Jonk2, 2023-02-18:
Hi Alex, I've installed a fresh Dev branch of allsky and have come here to fetch the extra modules.

The clone has worked, chmod'd the install script and ran it.

I get the error message:

`AllSky does not seem to be installed. The /opt/allsky directory does not exist. Please install AllSky before installing the modules`

Allsky is installed, so what do I need to check?

[image]

Jonk2, 2023-02-18:
Also, I'm not sure if this is related, but I've just noticed this, and I have no overlays showing...

[image]

Alex-developer, 2023-02-19:
@Jonk2 

I have moved all of the modules to /opt/allsky/modules You can just move the folder from /etc to /opt

You are getting the error because the /opt/allsky folder isnt there

Jonk2, 2023-02-19:
Ok, I've jumped through some hoops (ftp with root) and copied etc/allsky.... to opt/allsky

I've run the install script again, and it's stopped with this error:

```
Installing allsky_pistatus
==========================
INFO: Installing dependencies
INFO: Installing allsky_pistatus module
cp: cannot create regular file '/opt/allsky/modules/allsky_pistatus.py': Permission denied
ERROR: Installation of allsky_pistatus Failed, Could not copy module from /home/pi/allsky-modules/allsky_pistatus/allsky_pistatus.py to /opt/allsky/modules
```
Do I need to run sudo ./install.sh then? I'll try that now.

Jonk2, 2023-02-19:
The answer is no!

`DO NOT run this as root. Run the installer as the same user as AllSky was installed`

Alex-developer, 2023-02-19:
Permissions are very important. The permissions on the folder in /opt must be the same as they are in /etc

Change the permissions on the /opt/allsky folder (and all folder/files within) to the user you installed allsky as and www-date

so something like 

chown -R pi:www-data /opt/allsky

Jonk2, 2023-02-19:
Doesn't seem to work:

```
pi@rorpi:~/allsky-modules $ chown -R pi:www-data /opt/allsky
chown: changing ownership of '/opt/allsky/modules': Operation not permitted
chown: changing ownership of '/opt/allsky': Operation not permitted

```

etc/allsky is pi www-data

[image]

Jonk2, 2023-02-19:
Success! Thanks for your help.

[image]

Jonk2, 2023-02-19:
Ok I'm confused. Does this automatically now show in the module editor (I have rebooted after installation successful) or do I need to do something else?

Alex-developer, 2023-02-19:
It will appear in the module editor but only in the periodic flow

Jonk2, 2023-02-19:
Sorry, not for me it doesn't!

[image]

Alex-developer, 2023-02-19:
Doesnt for me either on a fresh install - leave it with me

Alex-developer, 2023-02-19:
In the module settings make sure 'Show experimental' is enabled

Jonk2, 2023-02-20:
It is, probably by default as I don't recall changing anything?

[image]

Jonk2, 2023-02-20:
Also there's a typo on that screen "Show exterimental modules". Is that external experimental?!

closed 2023-05-31 by Alex-developer