AllskyTeam/allsky-modules #8: Enhancement - DewHeater add pressure as usable system variable

jcauthen78, 2023-07-10:
Scenario:  Using the BME280 chip with the Dew script works well and provides the system variables, but doesn't provide the chip's available Pressure value.  This makes the user require a second script to query the BME280 in order to pull this variable - on my system having this exact setup, the Dew & Secondary script can both attempt to run at the same time causing one of them to error out because it can't find the chip because it's in use. 

Is it possible to pull/store the value for use in the overlay editor, even if its not necessarily needed for the dew function?

Alex-developer, 2023-07-26:
Working on this now Jim. I will add all of the values from all of the sensors. Most only return temp and humidity but the 280 has a few others like pressure and altitude

Alex-developer, 2023-07-27:
@jcauthen78 Module updated to include  pressure and altitude

closed 2023-08-23 by Alex-developer