AllskyTeam/allsky-modules #26: InfluxDB and Grafana setup

heruan, 2023-10-26:
I've read [this conversation](https://github.com/thomasjacquin/allsky/discussions/815) and I'd be really interested to replicate a similar Grafana dashboard for [our AllSky setup](http://astrofilivicentini.ddns.net/allsky/)! The discussion is thorough but I miss some starting points, e.g. which modules to use (if avaialble) or what is needed to install on the Raspberry and ultimately if the dashboard is anywhere available to download.

Is a guide available to create such setup?

Alex-developer, 2023-10-26:
The dashboard is one I created in Grafana for my own use, I am in the process of rebuilding it and will happily share it once done.

 For now it needs the following installed
 
 - Allsky !
 - The Extra modules from this repo
 - Influxdb setting up
 - The influxdb module setting up
 - Something like Grafana to display the data

heruan, 2023-10-27:
Thanks! I installed InfluxDB and the module on my Raspberry Pi 4. Using `apt install influxdb` installed version 1.6.7
On the module I have configure these settings:
[image]

but when saving the settings I get this error:
[image]

If I check the checkbox, then I can save but on the logs I get:
```
Oct 28 00:10:24 allsky allsky[17307]: ERROR:influxdb_client.client.write_api:The batch item wasn't processed successfully because: (404)
Oct 28 00:10:24 allsky allsky[17307]: Reason: Not Found
Oct 28 00:10:24 allsky allsky[17307]: HTTP response headers: HTTPHeaderDict({'Content-Type': 'text/plain; charset=utf-8', 'X-Content-Type-Options': 'nosniff', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': '1.6.7~rc0', 'Date': 'Fri, 27 Oct 2023 22:10:24 GMT', 'Content-Length': '19'})
Oct 28 00:10:24 allsky allsky[17307]: HTTP response body: 404 page not found
```

Looks like a validation issue on the module config blocking me to configure the module for InfluxDB 1.x

heruan, 2023-10-28:
Okay I fixed the 404 issue upgrading InfluxDB to 1.8 (by downloading the binaries from their GitHub repo).

Still, I get the validation error when saving module options with the `v2 Bucket` checkbox not checked, so I had to manually change the JSON file for that but every time I make changes it now complains, that would be great to be fixed!