AllskyTeam/allsky-modules #27: Allsky modules (Export, Clear Sky) cannot be selected and saved

WirthmU, 2023-10-27:
Neither the Export nor the Clear Sky modules can be activated and saved in my configuration. Also the settings are not accessible.

Alex-developer, 2023-11-01:
What happens when you try to activate them?

WirthmU, 2023-11-01:
I click on the activation checkbox
The checkbox sign appears
I click on the settings box nothing nothing happens
I save the module by clicking on the green sign icon which turns black
When I leave the modules section and return afterwards the check is is no longer activated

Alex-developer, 2023-11-01:
> I click on the activation checkbox The checkbox sign appears I click on the settings box nothing nothing happens I save the module by clicking on the green sign icon which turns black When I leave the modules section and return afterwards the checkbox is no longer activated

Do you know howto use the browser dev tools? If you open the dev tools and look at he 'console' tab are there any errors display whilst you are trying to get to the settings?

WirthmU, 2023-11-02:
This is what it looks like:

[image]

Alex-developer, 2023-11-02:
Those errors are nothing to worry about and won't be causing the issue you have. To be honest I am a little stumped by this one. Its not something I have seen before.

I'll have a think about it !

WirthmU, 2023-11-03:
Could it be a permission issue? Where are the files located?

Alex-developer, 2023-11-05:
> Could it be a permission issue? Where are the files located?

Could be they are all in the ~/allsky/config/modules folder.

WirthmU, 2023-11-06:
This is how the folder looks like:

[image]

It seems that there are no modules at all!

Alex-developer, 2023-11-06:
> This is how the folder looks like:
> 
> [image]
> 
> It seems that there are no modules at all!

They look fine. These are the files that store the module settings and if the permissions on these were wrong it would fail to save but they are fine

Still stumped :-(

WirthmU, 2023-11-09:
The problem has been solved. It was an error on my side. I should have read the manual where its clearly written that to activate any module it has to be dragged to the active ones, which I did not.
Thank you for your support!

closed 2023-11-09 by WirthmU