AllskyTeam/allsky-modules #39: module allsky_dewheater expansion with the “sensor” to include Solo Cloudwachter

Alex-developer, 2023-12-22:
### Discussed in https://github.com/AllskyTeam/allsky-modules/discussions/38

<div type='discussions-op-text'>

<sup>Originally posted by **Adler6907** December 18, 2023</sup>
I have expanded the `allsky_dewheater` module to include a “sensor”. The new "sensor" reads the weather data from the weather station "Solo Cloudwatcher" from the company "Lunatico Astronomia" (lunaticoastro.com) via HTTP. 

Runs on an Rpi5 with Bookwoorm.
new libraries `import gpiod`,  `import urllib.request` and `import json` are needed (they were already installed on mine)
new function `def readSolo(url)`, Changes in `metaData`, `getSensorReading` and `dewheater`

[image]

Here is the data provided by the “Solo Cloudwatcher”:
`{ "LastReadings": {
"dataGMTTime" : "2023/12/18 18:22:07",
"cwinfo" : "Serial: 2550, FW: 5.89",
"clouds" : -18.130000,
"cloudsSafe" : "Safe",
"temp" : 7.820000,
"wind" : 7,
"windSafe" : "Safe",
"gust" : 8,
"rain" : 3100,
"rainSafe" : "Safe",
"lightmpsas" : 19.82,
"lightSafe" : "Safe",
"switch" : 1,
"safe" : 1,
"hum" : 40,
"humSafe" : "Safe",
"dewp" : -4.940000,
"rawir" : -22.680000,
"abspress" : 998.450000,
"relpress" : 1027.527780,
"pressureSafe" : "Safe"
}
}`

Here in the zip file is also another module "allsky_solo_cloudwatcher" which writes all the data listed above from the "Solo-Cloudwatcher" to `/home/pi/allsky/tmp/allskydb.py` for possible use in other modules

[allsky_dewheater.zip](https://github.com/AllskyTeam/allsky-modules/files/13728925/allsky_dewheater.zip)

CS</div>

Alex-developer, 2023-12-23:
@Adler6907 There is now a version of the dew heater in the dev branch with the Solo code in it.

I have refactored the code a little. Please can you give it a test

Thanks

Alex

Adler6907, 2023-12-24:
I have tested it. First I had to install the library manually because of `import adafruit_ahtx0` with
`(venv) pi@allsky1: ~ $ pip3 install adafruit-circuit-ahtx0`

It is not yet listed in `/home/pi/allky module/allsky_dewheater/requirements.txt`. Or is that in the standard in `Allsky/Config_Repo/Requirements.txt`?

In "allsky_dewheater.py" there is still a small mistake in the `readSolo(url)` function.

The lines
```
     try: 
        #Read Weaterdata from SOLO Website
        jsonData = urllib.request.urlopen(url).read()  
        response = json.dumps(json.loads(jsonData)["LastReadings"]) 
        
        #response = '{ "LastReadings": { "dataGMTTime" : "2023/12/18 18:22:07", "cwinfo" : "Serial: 2550, FW: 5.89", "clouds" : -18.130000, "cloudsSafe" : "Safe", "temp" : 7.820000, "wind" : 7, "windSafe" : "Safe", "gust" : 8, "rain" : 3100, "rainSafe" : "Safe", "lightmpsas" : 19.82, "lightSafe" : "Safe", "switch" : 1, "safe" : 1, "hum" : 40, "humSafe" : "Safe", "dewp" : -4.940000, "rawir" : -22.680000, "abspress" : 998.450000, "relpress" : 1027.527780, "pressureSafe" : "Safe" } }'
        currentWeatherdata =  json.loads(response)['LastReadings']

        # that is what you should receive
```
should be changed in this way. Then it works.

```
        #Read Weaterdata from SOLO Website
        jsonData = urllib.request.urlopen(url).read()  

        currentWeatherdata =  json.loads(jsonData)['LastReadings']
        # that is what you should receive in jsonData 
```

`response` is unnecessary

Alex-developer, 2023-12-25:
> I have tested it. First I had to install the library manually because of `import adafruit_ahtx0` with `(venv) pi@allsky1: ~ $ pip3 install adafruit-circuit-ahtx0`
>
> It is not yet listed in [...]

Thanks for that.

The missing module was my mistake - I have added the i2c DHT22. I have now added the module to the requirements.

I have changed the Solo code. If you could please give it another test that would be appreciated.

Thanks

Alex

Adler6907, 2023-12-26:
It could be that the incorrect code in “readSolo” came from me. Whatever the reason, I can no longer understand it. But that's why we test ;-)

Now the module works perfectly with the SOLO as a “sensor”.

For "Adler6907" you can also use my name "Andreas Schminder" as with the others if you like.

The whole mechanism with the overlay modules, periodic execution, etc. is ingeniously done. That deserves respect. I take my hat off :-)

I'm currently creating a periodic module, which uses the wind weather data (Solo (wind strength and gusts) and Wunderground (wind direction)) to make an average of the last time the shutter is protected from the wind (it always raises and lowers when the wind is strong and the wind direction is unfavorable). Shelly relays are switched, the dome control and the MiniPC in the observatory are started up, the dome is rotated via the ASCOM Alpaca server with a curl command and everything is shut down again cleanly.

Alex-developer, 2023-12-26:
Andreas

I broke the code when refactoring it so my fault :-)

I have updated the contrib info with your name and merged it all into the master branch.

Glad you like the modules/overlay system. There is a lot more to come to make it even more flexible. Watch this space (Or just read the issues as its all in there)

I'll close this issue now as its all merged. Many thanks for the contribution

Alex

closed 2023-12-26 by Alex-developer