AllskyTeam/allsky-modules #52: Add custom path for overlay images

Alex-developer, 2023-12-29:
Add a custom path for users to save images in, this could be in a tmpfs

Also see https://github.com/AllskyTeam/allsky-modules/issues/40 as this needs to be done at the same time

ssagerian, 2023-12-29:
works for me. One other comment might be to add a way for the user to include an image in the System page. A png graph showing some kind of status, thus not having to clutter the live image.

Alex-developer, 2023-12-29:
I am working on a whole new web interface for Allsky part of which allows you to add custom images, data and controls. As an example I have added your graph

[image]