AllskyTeam/allsky-modules #80: No modules listed in module manager after installation of extra Allsky modules

OmniversePhotography, 2024-02-03:
**Describe the bug**
On a fresh setup of Allsky the default modules are all present in the module manager. After following the installation instructions and installing the extra modules via https://github.com/AllskyTeam/allsky-modules.git I get an empty list of modules in the module manager in the web UI. All python files are still found in both /opt/allsky/modules and /home/pi/allsky/scripts/modules/.

**To Reproduce**
Steps to reproduce the behaviour:
1. git clone https://github.com/AllskyTeam/allsky-modules.git to home directory on pi.
2. cd into cloned allsky-modules directory
3. run ./install.sh
4. after python venv starts navigate to "Install/Remove Modules"
5. Select all and press enter. Wait for install to finish
6. Restart Pi 
7. Go to module manager on webUI and the list is now empty. 

**Expected behaviour**
Expected to see additional modules listed alongside original default modules in the webUi manager.

**Screenshots**
I
[image]
[image]
[image]

**Allsky Information:**
 - OS: [Raspberry Pi OS, 64bit, lite, version 12 Bookworm]
 - Allsky Version: v2023.05.01_04
 - Pi model: Raspberry Pi 4 Model B 8GB
 - Camera: ZWO ASI224MC

**Additional context**
This was all done on a fresh installation of Raspberry Pi OS. As a test i wiped the SD card and ran through the setup again with the same outcome.

OmniversePhotography, 2024-02-03:
I will also add that I found someone else had posted this problem on this discussions page and it was recommended to delete /opt/allsky/modules/allsky_test.py. Before deleting the file i looked at the contents but it was empty. I proceeded to delete the file but unfortunately it didn't fix the problem.

OmniversePhotography, 2024-02-03:
Just an update. I wiped my Pi again and began reinstallation. This time when selecting which extra modules I wanted to install, I only selected the two I'm currently using (Dew Heater Module and Open Weather Map). Everything is working now as it should, so the problem could be related to the installation of any of the other modules.

Alex-developer, 2024-02-03:
Thanks for the report. I dont suppose you kept a copy of the logs files from the installation installation?

OmniversePhotography, 2024-02-04:
> Thanks for the report. I don't suppose you kept a copy of the logs files from the installation installation?

Sadly it slipped my mind to save the log before wiping. That would have been good to check, sorry bout that.

Alex-developer, 2024-02-04:
> > Thanks for the report. I don't suppose you kept a copy of the logs files from the installation installation?
> 
> Sadly it slipped my mind to save the log before wiping. That would have been good to check, sorry bout that.

No problem :-)

cgmiller76, 2024-02-25:
@Alex-developer - I am having this same issue - so I didn't know if I should add a comment here or start a new thread.  I still have my log - so I'll add it here.  When I installed, I selected all the modules EXCEPT allsky_test, but it appears it still installed it.  I tried removing just the /home/pi/allsky-modules/allsky_test folder (it had been installed with the allsky_test.py within).

- RPi Model 4B Revision 1.4 (8 GB)
- RPi HQ camera
- AllSky Version: v2023.05.01_04     

[moduleinstalldebug.log](https://github.com/AllskyTeam/allsky-modules/files/14396694/moduleinstalldebug.log)

Alex-developer, 2024-02-26:
I have added a few fixes and tested installing all modules - All seems good now.

Please let me know how you get on

Cheers

Alex

cgmiller76, 2024-02-26:
I removed the allsky-modules directory and reinstalled without issue. All modules are now visible. Thanks!

commit e4182e7b (2024-02-26): Merge pull request #105 from AllskyTeam/ltr390-fix

#80 Hotfix to fix meta data in ltr390 module
closed 2024-02-26 by Alex-developer