AllskyTeam/allsky-modules #86: allsky_discordsend.py fails if all webhook URL fields are not filled

trevor229, 2024-02-13:
**Describe the bug**
When utilizing only the timelapse posting feature of allsky_discordsend.py, and if only the timelapse webhook URL is populated, the script will fail and log `ERROR: Module allsky_discordsend.py failed on line 224 - can only concatenate str (not "NoneType") to str`

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to the settings of the DiscordSend module in the allsky web UI
2. Enable only the "Post timelapse videos" checkbox
3. Add a valid discord webhook URL to only the "The webhook url for Timelapses"
4. Ensure other posting options and webhook fields are disabled or blank
5. Run `./endOfNight.sh` or wait for the script to run normally
6. Check /var/log/allsky.log for the above error

**Expected behaviour**
The script should either warn the user to fill out all webhook fields or only check and concatenate webhook fields of enabled features. Instead, the script fails and logs to `/var/log/allsky.log`.

When all webhook fields are filled, and the timelapse post feature is enabled, the script completes successfully upon running `./endOfNight.sh`

**Allsky Information:**
 - OS: Raspbian GNU/Linux 11 (bullseye) lite | Linux 6.1.21-v7l+
 - Allsky Version: v2023.05.01_04
 - Pi model: Pi 4 4GB
 - Camera: RPi HQ with Arducam M25156H18

**Sensors**
SHT31

Alex-developer, 2024-02-13:
Thanks I'll fix this in the next release

trevor229, 2024-02-14:
After letting the software do its thing for a day, there might be something else going on here.

`cat /var/log/allsky.log | grep -a "discord"` shows me the same error, but when I manually run `./endOfNight.sh` it generates the keogram, startrail, timelapse, and uploads it to FTP and discord without any issue. Not quite sure what's going on now.

italic-r, 2024-06-03:
I can confirm this behavior here. I have all fields filled and only the keogram enabled, but it won't upload on its own at the end of the night; it will only upload when I initiate `endOfNight.sh` manually. It will upload individual frames any time it's scheduled to for night or day, but it won't upload the keogram. I don't want timelapses uploaded so I have not tried those.

Alex-developer, 2024-06-04:
I have found the issue and am working on a fix

Alex-developer, 2024-06-07:
Fix add to the modules. You will also need to edit a script in the main allsky folder

Edit ~/allsky/scripts/endOfNight.sh

and add the following libe

set -a

so that the file looks like this

italic-r, 2024-06-07:
I'll give this a try and see if it uploads in the morning. Thanks!

trevor229, 2024-06-15:
I made the changes above and replaced `allsky_discordsend.py` with the new version. At first things did not appear to work and possibly broke, but once I used the version of `endOfNight.sh` in the main branch and added `set -a` as opposed to the `v2024.xx.xx` branch things are working properly and webhook functionality works. It appears a few lines are missing from the `v2024.xx.xx `version which are needed for the current version to operate properly.

Thank you for looking into this!

italic-r, 2024-06-18:
I forgot to report that my problem has been solved with these changes. Thanks!