AllskyTeam/allsky-modules #91: Light module not working

WirthmU, 2024-02-15:
I have a TSL2561 sensor on the standard 0x29 address.

Unfortunately no jason-file is created:

[image]

Any testing I could do?

Alex-developer, 2024-02-17:
Just tried mine and its working fine so ..

Which PI and OS are you using? I am using a pi 4 and bullseye lite

Is there anything logged in /var/log/allskyperiodic.log ?

WirthmU, 2024-02-18:
I am using a Raspberry PI 4 Model B Rev. 1.2 with Bookworm.

The error message in the log-file is:
```
2024-02-18T16:42:02.881932+01:00 meteotux allskperiodic[255813]: ERROR: Module allsky_light.py failed on line 224 - Failed to find TSL2561! Part 0x0 Rev 0x0
2024-02-18T16:43:03.648465+01:00 meteotux allskperiodic[255924]: ERROR: Module allsky_light.py failed on line 224 - Failed to find TSL2561! Part 0x0 Rev 0x0
2024-02-18T16:44:04.427703+01:00 meteotux allskperiodic[256034]: ERROR: Module allsky_light.py failed on line 224 - Failed to find TSL2561! Part 0x0 Rev 0x0
```

Alex-developer, 2024-02-18:
Looks like you may have the wrong i2c address in the module.

The 2561 has 3 possible 12c addresses 0x29, 0x39 and 0x49. The default is 0x39 which the module uses, it says it uses 0x29 but thats a typo which I will fix.

Put the i2c address of your 2561 in the module and see if that works.

WirthmU, 2024-02-18:
I did set the address in the settings to 0x29, saved and did a re-boot.
Strange enough the log shows:
`2024-02-18T20:13:29.392199+01:00 meteotux allskperiodic[948]: ERROR: Module allsky_light.py failed on line 224 - No I2C device at address: 0x39`

Alex-developer, 2024-02-19:
Hmm wierd

Whats the output of 

``i2cdetect -y -a 1``

If you don't have i2cdetect installed you can install it with the following

``sudo apt-get install i2c-tools``

WirthmU, 2024-02-20:
```
pi@meteotux:~ $ i2cdetect -y -a 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 -- -- -- 04 -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- 29 -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- 5b -- -- -- -- 
60: 60 -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
```

Alex-developer, 2024-02-20:
> ```
> pi@meteotux:~ $ i2cdetect -y -a 1
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00: 00 -- -- -- 04 -- -- -- -- -- -- -- -- -- -- --
> 10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- [...]

Thanks  - I have found the problem and will add a fix

Alex-developer, 2024-02-20:
I have added a fix for this. Can you try the new module, you will have to enter 0x29 as the i2c address

Thanks

Alex

WirthmU, 2024-02-23:
With the new version it is working now - that's great!

```
(foobar) pi@meteotux:~/allsky-modules $ cat /home/pi/allsky/config/overlay/extra/allskylight.json
{
    "AS_LIGHTLUX": "20.64502756305084",
    "AS_LIGHTNELM": "-5.391135777781152",
    "AS_LIGHTSQM": "8.263573974337595"
```

closed 2024-02-23 by WirthmU