AllskyTeam/allsky-modules #103: Adding the MLX90615 sensor to the cloud cover module

WirthmU, 2024-02-26:
The module states the MLX90614 sensor - would it be possible to use an MLX90615 on I2c port 0x5b?

Alex-developer, 2024-06-07:
I dont have a mlx90615 - If I add the code would you be willing to test it for me?

WirthmU, 2024-06-10:
I am currently running Allsky-Indi on the RPI with the MLX90615 Sensor. I have deactivated this and did a fresh install of Allksy.
Unfortunately there seems to be a problem with lightppd and Allsky does not start:
```
(foobar) pi@meteotux:~ $ systemctl status lightppd
Unit lightppd.service could not be found.
(foobar) pi@meteotux:~ $ service lighttpd status
× lighttpd.service - Lighttpd Daemon
     Loaded: loaded (/lib/systemd/system/lighttpd.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Sun 2024-06-09 11:42:00 CEST; 19min ago
   Duration: 393ms
    Process: 1141 ExecStartPre=/usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf (code=exited, status=0/SUCCESS)
    Process: 1146 ExecStart=/usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf (code=exited, status=255/EXCEPTION)
   Main PID: 1146 (code=exited, status=255/EXCEPTION)
        CPU: 788ms

Jun 09 11:42:00 meteotux systemd[1]: lighttpd.service: Scheduled restart job, restart counter is at 5.
Jun 09 11:42:00 meteotux systemd[1]: Stopped lighttpd.service - Lighttpd Daemon.
Jun 09 11:42:00 meteotux systemd[1]: lighttpd.service: Start request repeated too quickly.
Jun 09 11:42:00 meteotux systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Jun 09 11:42:00 meteotux systemd[1]: Failed to start lighttpd.service - Lighttpd Daemon.
(foobar) pi@meteotux:~ $ lighttpd -t -f /etc/lighttpd/lighttpd.conf
```

Alex-developer, 2024-06-10:
Is there anything in /var/log/lighttpd/error.log?

WirthmU, 2024-06-17:
/var/log/lighttpd/error.log is empty

Alex-developer, 2024-06-17:
Try running the following command and se if it produces any errors

/usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf

WirthmU, 2024-06-18:
No errors:
```
(foobar) pi@meteotux:~ $ /usr/sbin/lighttpd -tt -f /etc/lighttpd/lighttpd.conf
(foobar) pi@meteotux:~ $ 
```

closed 2024-11-20 by Alex-developer