AllskyTeam/allsky-modules #112: Add -clean folder to the webui

Alex-developer, 2024-03-07:
### Discussed in https://github.com/AllskyTeam/allsky-modules/discussions/111

<div type='discussions-op-text'>

<sup>Originally posted by **custard2022** March  6, 2024</sup>
Hello,
Trying to figure out why the -clean folders exist and cannot be seen or removed using the WebUI.

In Allsky folder:
[image]
In WebUI: 
[image]

I had it set to Daytime Capture, but __NOT__ Daytime Save, the folders and images get saved, but cannot be removed or seen in the WebUI, taking up space unless manually deleted.

My question is:
1. What option does it make them save to file, and how to remove them?
2. What option is there to NOT save them, but only preview the camera Live without saving thousands of pictures?</div>

NiteRide, 2025-08-14:
@EricClaeys -
Maybe can tie this in with v2025 as part of the change to show the test folders in the WebUI?

EricClaeys, 2025-08-14:
@NiteRide, LOL, I don't remember every seeing this Issue before.

@Alex-developer,
the "saveintermediateimage" module creates these directories and stores the intermediate images there?
* Should these directories be deleted when the "Days to Keep" check is done at the end of night?
* Should they appear in the WebUI's "Images" page?