AllskyTeam/allsky-modules #120: Module allsky_publishdata.py failed on MQTT data publication

PiotrNomad, 2024-03-22:
**Describe the bug**
When trying to publish data to MQTT broker the module throws the exception in allsky.log:
ERROR: Module allsky_publishdata.py failed on line 224 - Client.__init__() missing 1 required positional argument: 'callback_api_version'
Nothing beeing published to MQTT.
 
**To Reproduce**
Steps to reproduce the behaviour:
1. install BookWorm 32 bit lite on Rasberry pi
2. install Allsky
3. install publishdata module
4. configure MQTT broker
5. Activate publishdata module 
6. Restrt Allsky
7. Waith for the 1st picture publication
8. See error in /var/logs/allsky.log

**Expected behaviour**
No error in log
data been published

**Allsky Information:**
 - OS: Linux allsky 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 
 - Allsky Version: 2023.05.01_04
 - Pi model: Model 4B Revision 1.4 (2 GB)
 - Camera: HQ

**Additional context**
It seams thst this is typical behavior on BookWorm Rasbian.
I had the same issue with my code and solved it by: client = paho.Client(paho.CallbackAPIVersion.VERSION1)

PiotrNomad, 2024-03-24:
Some resources: 
https://eclipse.dev/paho/files/paho.mqtt.python/html/migrations.html
https://stackoverflow.com/questions/77984857/paho-mqtt-unsupported-callback-api-version-error

PiotrNomad, 2024-03-24:
When I change theline 254 from:
client = paho.Client(client_id="", userdata=None, protocol=paho.MQTTv5)
to:
client = paho.Client(paho.CallbackAPIVersion.VERSION1, client_id="", userdata=None, protocol=paho.MQTTv5)
it works :)

Alex-developer, 2024-03-24:
Thanks I'll take a look once I have finished testing the next release of Allsky