AllskyTeam/allsky-modules #129: Right Data Type in MQTT Data Publish

starhopper-de, 2024-04-23:
I use the MQTT Data Publish Module to export environment data to a remote server. It has a influxDB to store and Grafana to visualize this data. 

**Is your feature request related to a problem? Please describe.**

I had to place a NODE-RED instance in front of this because numeric date come as string.
Example: "AS_DEWCONTROLAMBIENT": "23.7"

**Describe the solution you'd like**

Is it possible to provide numeric as numeric, string as string and boolean as boolean.

Thomas

Alex-developer, 2024-06-07:
Yes, its a great idea

I'll add an option to convert any numeric/boolean values and leave others as strings

Alex-developer, 2024-06-08:
@starhopper-de I will merge a change that will set the correct data types

closed 2024-06-08 by Alex-developer