AllskyTeam/allsky-modules #135: Bug module cloud

ettoreribaudo, 2024-05-12:
**Describe the bug** random i have this error: ERROR: Module cloud failed on line 251 - module 'board' has no attribute 'I2C'

bus I2c Rasperry is ok.

**To Reproduce**

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Allsky Information:**
 - OS: [Bullseye 64bit and Full]
 - Allsky Version: last
 - Pi model: 4b 4 gb ram
 - Camera: innomaker 462

**Sensors**
MLX90614

[image]
[image]

ettoreribaudo, 2024-05-18:
Please responde .
Ettore

Alex-developer, 2024-06-07:
Closing as I am responding in the discussion

closed 2024-06-07 by Alex-developer