AllskyTeam/allsky-modules #137: i2c address selection of SHT31D sensor for dewheater & temp modules

mgarza, 2024-05-28:
It would be great if there is a chance to allow different I2C addresses for the SHT31D for the dewheater and temp modules.
For some other sensors this functionality seems to be implemented, but it's missing for the SHT31D.

e.g.: 
....
`def readSHT31(sht31heater,i2caddress):`
`    temperature = None`
`    humidity = None`
`    if i2caddress != "":`
`        try:`
`            i2caddressInt = int(i2caddress, 16)`
`        except Exception as e:`
`            eType, eObject, eTraceback = sys.exc_info()`
`            s.log(0, f"ERROR: Module readSHT31 failed on line {eTraceback.tb_lineno} - {e}")`
`    try:`
`        i2c = board.I2C()`
`        if i2caddress != "":`
`            sensor = adafruit_sht31d.SHT31D(i2c,i2caddressInt)`
`        else:`
`            sensor = adafruit_sht31d.SHT31D(i2c)`
....

Alex-developer, 2024-06-04:
Thanks for the idea, I'll add the option

Alex-developer, 2024-06-07:
I have added the ability to set the i2c address for the SHT31 to both the temp and dewheater module. YOu will need to reinstall the modules to get the new functionality

closed 2024-06-07 by Alex-developer

mgarza, 2024-06-08:
Thank you very much for your fast support :-)