AllskyTeam/allsky-modules #138: module allskyAi

ettoreribaudo, 2024-05-31:
**Describe the bug**
Last week it worked but now it always gives me fake but the photos are uploaded to the allskyAI.com website

I uninstalled and reinstalled the module but still false

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Allsky Information:**
 - OS: Bullseye
 - Allsky Version: last
 - Pi model: PI4 B 4 GB
 - Camera: imx462 PI
**Sensors**
[image]

ettoreribaudo, 2024-06-02:
Nothing response?

Alex-developer, 2024-06-04:
@TheCrowsNestAstro Can you shed any light on this issue?

ettoreribaudo, 2024-06-06:
This web is blocked: https://www.allskyai.com/

Alex-developer, 2024-06-06:
I have spoken to the developer of allskyai and he is looking into the website issue

ettoreribaudo, 2024-06-07:
Web allskyai.com is ok.

Alex-developer, 2024-06-07:
Online for me as well - Is the module working now?

ettoreribaudo, 2024-06-07:
Yes perfect.

Alex-developer, 2024-06-07:
Thanks will close the issue

closed 2024-06-07 by Alex-developer

ettoreribaudo, 2024-07-08:
Hello Alex same problems module it is no longer found in the variables.
Ettore
[image]
[image]