AllskyTeam/allsky-modules #157: Problem displaying new modules on overlay

Kilis71, 2024-08-14:
I installed the external modules but when I activate for example the weather forecast module, I don't see it in the overlay, what should I do?

Alex-developer, 2024-08-17:
> I installed the external modules but when I activate for example the weather forecast module, I don't see it in the overlay, what should I do?

Have you looked over the documentation on howto add variables?

Most of the modules provide variables that you can use in the overlay editor

Kilis71, 2024-08-18:
Of course, it's precisely in the overlay editor that I don't see them.

Kilis71, 2024-08-18:
I have installed additional modules, the temperature, humidity work and I display them, while the rain sensor does not work, the module practically does not create the .json file in the extra folder and does not create the variables to put on top, same thing for open weather map

Kilis71, 2024-08-18:
Could the malfunction of the rain detector be the fault of the GPIO which seems to be managed differently in the Raspberry 5 than in the Raspberry 4? I'm using an RPi5

Alex-developer, 2024-08-19:
In the main allsky settings set the debug level to 4 and have a look in the following files and see if any errors are being generated

/var/log/allsky.log
/var/log/allskyperiodic.log

Cheers

Alex

Kilis71, 2024-08-19:
in this log (/var/log/allskyperiodic.log)  it gives me an error

2024-08-19T19:16:11.938687+02:00 allsky allskperiodic[98714]: INFO: ----------------------- Running Module allsky_temp.py -----------------------
2024-08-19T19:16:11.938722+02:00 allsky allskperiodic[98714]: INFO: Attempting to load allsky_temp
2024-08-19T19:16:11.938747+02:00 allsky allskperiodic[98714]: INFO: Reading sensor 1, SHT31
2024-08-19T19:16:11.938775+02:00 allsky allskperiodic[98714]: INFO: Sensor SHT31 read. Temperature 27.19 Humidity 65.2 Relative Humidity None Dew Point 20.09 Heat Index -5.18 Pressure None Altitude None
2024-08-19T19:16:11.938802+02:00 allsky allskperiodic[98714]: INFO: Module allsky_temp.py ran ok in 0.04s
2024-08-19T19:16:11.938825+02:00 allsky allskperiodic[98714]: INFO: ----------------------- Running Module allsky_rain.py -----------------------
2024-08-19T19:16:11.938935+02:00 allsky allskperiodic[98714]: INFO: Attempting to load allsky_rain
**2024-08-19T19:16:11.938996+02:00 allsky allskperiodic[98714]: ERROR: Unable to read Rain sensor Cannot determine SOC peripheral base address**
2024-08-19T19:16:11.939020+02:00 allsky allskperiodic[98714]: INFO: Module allsky_rain.py ran ok in 0.00s
2024-08-19T19:16:11.939039+02:00 allsky allskperiodic[98714]: INFO: ----------------------- Running Module allsky_openweathermap.py -----------------------
2024-08-19T19:16:11.939067+02:00 allsky allskperiodic[98714]: INFO: Attempting to load allsky_openweathermap
2024-08-19T19:16:11.939090+02:00 allsky allskperiodic[98714]: INFO: Last run 242.6600308418274 seconds ago. Running every 600 seconds
2024-08-19T19:16:11.939116+02:00 allsky allskperiodic[98714]: INFO: Module allsky_openweathermap.py ran ok in 0.00s
2024-08-19T19:16:11.939276+02:00 allsky allskperiodic[98714]: INFO: ----------------------- Running Module allsky_pistatus.py -----------------------
2024-08-19T19:16:11.939337+02:00 allsky allskperiodic[98714]: INFO: Attempting to load allsky_pistatus
2024-08-19T19:16:11.939385+02:00 allsky allskperiodic[98714]: INFO: PI Status Data Written
2024-08-19T19:16:11.939467+02:00 allsky allskperiodic[98714]: INFO: Module allsky_pistatus.py ran ok in 0.12s
2024-08-19T19:16:11.939532+02:00 allsky allskperiodic[98714]: INFO: periodic flow Complete...
2024-08-19T19:16:12.088382+02:00 allsky allskperiodic[3773]: INFO: Sleeping for 60 seconds

Kilis71, 2024-08-19:
Fixed the weather issue, it was an error in the API key. the problem of the rain sensor remains

Kilis71, 2024-08-19:
I solved the rain sensor problem. on raspberry 5 the GPIO ports are managed differently, you need to uninstall python3-rpi.gpio and install python3-rpi-lgpio, it works now. the commands are as follows:
`sudo apt remove python3-rpi.gpio`
`sudo apt install python3-rpi-lgpio`

Alex-developer, 2024-08-28:
Is this a recent install of Allsky? The current release should install those libraries for you. I am curious as to why they were not there

Cheers

Alex

Kilis71, 2024-08-29:
Yes, Installing with latest OS version on RPi5 and latest version of allskycam. 🤷

Alex-developer, 2024-08-29:
> Yes, Installing with latest OS version on RPi5 and latest version of allskycam. 🤷

That is indeed odd. I'll run some tests as I need to rebuild one of my dev cameras 

Alex

Alex-developer, 2024-09-17:
There has been a kernel update that has caused issues with gpio in some areas. I am looking to see if this is the same issue, it does appear to be