AllskyTeam/allsky-modules #163: allsky_discordsend do not send images with overlay datas

drakula1981, 2024-09-16:
I've Activated the discord send module. 
It works fine and send images at the setted interval.
BUT, when I create images with the legacy overlay datas manager, the final image contains the overlays datas and when I activate the module manager, the image is sent raw, without any overlay datas.

**To Reproduce**
Steps to reproduce the behaviour:
1. Activate the Discord send module and configure it to be able to send daytime and nighttime images to a discord channel
2. Activate the legacy overlay datas manager, setup a legacy data and wait for an image sent to the discord channel, the overlay datas are visible
3. Switch to the module overlay data manager, put some overlay text in the overlay editor, save it and wait for an image sent to the discord channel, the overlay datas aren't visible

**Expected behaviour**
The overlayed image is sent to the channel when the module overlay manager is used

**Screenshots**
With the legacy overlay manager :
[image]
With the module overlay manager :
[image]

**Allsky Information:**
 - OS: Bookworm, 64bits lite up to date
 - Allsky Version: v2023.05.01_04
 - Pi model: RaspBerry Pi 5 4GB
 - Camera: Zwo ASI678MC 
 
**Sensors**
BME280 on I2C Address 0x77

**Additional context**
N/A

drakula1981, 2024-09-17:
I took a look at the allsky_discordsend.py file and I saw this :
[image]
could it be the part of the problem ?
Thus, I thought that the modules are arranged In the module manager, i put the "discord send" module right after the "overlay datas fields on the image" module as seen in the screenshot below :
[image]
But it has no effects

Alex-developer, 2024-09-17:
This was a deliberate decision by me as I wanted the clean images in Discord.

I'll add an option to send either the clan image or the annotated image

Cheers

Alex

Alex-developer, 2024-09-17:
Code all done, I am testing it on my pi just to make sure that the end of night discord sends still work. Once i am happy I will merge the code.

If you want to try it before hand its in the 163- branch

Alex

drakula1981, 2024-09-17:
Hi alex!
Well done!
I´ll give a try on the next days and give you my feedback!
Thanks for your reactivity!

Alex-developer, 2024-09-18:
This work overnight for my setup so have merged the change into master. If you reinstall the discord module there is an option against day and night to either send the annotated or unannotated image.

closed 2024-09-18 by Alex-developer