AllskyTeam/allsky-modules #192: GPIO control error on RPi5

chvvkumar, 2024-12-09:
**Describe the bug**
Currently trying the dewcontrol module and I am seeing the following in the logs. 

**To Reproduce**
Steps to reproduce the behaviour:
1. Use BME280 sensor
3. Use GPIO to control relay for dew heater

**Expected behaviour**
GPIO pin should be controlled from within the script

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Allsky Information:**
 - OS: Debian GNU/Linux 12 (bookworm), Raspberry Pi OS Lite 64-bit
 - Allsky Version: v2024.12.06
 - Pi model: Raspberry Pi 5 Model B Rev 1.0, 4GB
 - Camera: HQ Camera

**Sensors**
BME280 - I2C

```
pi@allskypi5:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- 76 --
```

**Additional context**

Error:
```
2024-12-08T18:24:03.591750-06:00 allskypi5 allskperiodic[3148]: INFO: Loading /home/pi/allsky/config/settings.json
2024-12-08T18:24:03.591918-06:00 allskypi5 allskperiodic[3148]: INFO: ===== Running periodic flow...
2024-12-08T18:24:03.591936-06:00 allskypi5 allskperiodic[3148]: INFO: --------------- Running Module allsky_dewheater ---------------
2024-12-08T18:24:03.591952-06:00 allskypi5 allskperiodic[3148]: INFO: Temperature outside limit temperature 15.83, limit 10, dewPoint -1.08
2024-12-08T18:24:03.591973-06:00 allskypi5 allskperiodic[3148]: ERROR: Module allsky_dewheater.py failed on line 209 - Cannot determine SOC peripheral base address
2024-12-08T18:24:03.591997-06:00 allskypi5 allskperiodic[3148]: INFO: --------------- Running Module allsky_pistatus ---------------
2024-12-08T18:24:03.592014-06:00 allskypi5 allskperiodic[3148]: INFO: PI Status Data Written
2024-12-08T18:24:03.592039-06:00 allskypi5 allskperiodic[3148]: INFO: --------------- Running Module allsky_publishdata ---------------
2024-12-08T18:24:03.592054-06:00 allskypi5 allskperiodic[3148]: ERROR: Cannot locate environment variable AS_DEWCONTROLAMBIENT specified in the extradata
2024-12-08T18:24:03.592071-06:00 allskypi5 allskperiodic[3148]: ERROR: Cannot locate environment variable AS_DEWCONTROLHUMIDITY specified in the extradata
2024-12-08T18:24:03.592088-06:00 allskypi5 allskperiodic[3148]: ERROR: Cannot locate environment variable AS_DEWCONTROLDEW specified in the extradata
2024-12-08T18:24:03.592102-06:00 allskypi5 allskperiodic[3148]: ERROR: Cannot locate environment variable AS_DEWCONTROLHEATER specified in the extradata
2024-12-08T18:24:03.592116-06:00 allskypi5 allskperiodic[3148]: INFO: MQTT - Published to MQTT on channel: Astro/AllSky
2024-12-08T18:24:03.592132-06:00 allskypi5 allskperiodic[3148]: INFO: MQTT - CONNACK received with code Success.
2024-12-08T18:24:03.592146-06:00 allskypi5 allskperiodic[3148]: INFO: ===== periodic flow complete.
2024-12-08T18:24:03.740243-06:00 allskypi5 allskperiodic[2865]: INFO: Sleeping for 60 seconds
```

Additional OS details:

```
pi@allskypi5:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
pi@allskypi5:~ $ hostnamectl
 Static hostname: allskypi5
       Icon name: computer
      Machine ID: bf352f9be0e04948b56f17b00a27d47e
         Boot ID: 5563076c675945f4a6e474975d5f7012
Operating System: Debian GNU/Linux 12 (bookworm)
          Kernel: Linux 6.6.62+rpt-rpi-2712
    Architecture: arm64

```

Troubleshooting steps so far:

Followed this answer to see if installing the new Pi5 libraries would help, but it did not make any difference.
https://github.com/AllskyTeam/allsky-modules/issues/157#issuecomment-2297612853

```
pi@allskypi5:~ $ sudo apt remove python3-rpi.gpio
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package 'python3-rpi.gpio' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  edid-decode libqt5qmlworkerscript5 libqt5quickcontrols2-5 libqt5quicktemplates2-5 linux-headers-6.6.31+rpt-common-rpi linux-headers-6.6.31+rpt-rpi-2712
  linux-headers-6.6.31+rpt-rpi-v8 linux-image-6.6.31+rpt-rpi-2712 linux-image-6.6.31+rpt-rpi-v8 linux-kbuild-6.6.31+rpt python3-gi-cairo qml-module-qtgraphicaleffects
  qml-module-qtquick-controls2 qml-module-qtquick-layouts qml-module-qtquick-templates2 qml-module-qtquick-window2 qml-module-qtquick2 rpi.gpio-common xinput
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

pi@allskypi5:~ $ sudo apt install python3-rpi-lgpio
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  edid-decode libqt5qmlworkerscript5 libqt5quickcontrols2-5 libqt5quicktemplates2-5 linux-headers-6.6.31+rpt-common-rpi linux-headers-6.6.31+rpt-rpi-2712
  linux-headers-6.6.31+rpt-rpi-v8 linux-image-6.6.31+rpt-rpi-2712 linux-image-6.6.31+rpt-rpi-v8 linux-kbuild-6.6.31+rpt python3-gi-cairo qml-module-qtgraphicaleffects
  qml-module-qtquick-controls2 qml-module-qtquick-layouts qml-module-qtquick-templates2 qml-module-qtquick-window2 qml-module-qtquick2 rpi.gpio-common xinput
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  python3-rpi-lgpio
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 14.3 kB of archives.
After this operation, 59.4 kB of additional disk space will be used.
Get:1 http://archive.raspberrypi.com/debian bookworm/main arm64 python3-rpi-lgpio all 0.6-0~rpt1 [14.3 kB]
Fetched 14.3 kB in 0s (30.7 kB/s)
Selecting previously unselected package python3-rpi-lgpio.
(Reading database ... 210576 files and directories currently installed.)
Preparing to unpack .../python3-rpi-lgpio_0.6-0~rpt1_all.deb ...
Unpacking python3-rpi-lgpio (0.6-0~rpt1) ...
Setting up python3-rpi-lgpio (0.6-0~rpt1) ...
pi@allskypi5:~ $ sudo reboot

Broadcast message from root@allskypi5 on pts/1 (Sun 2024-12-08 18:19:25 CST):

The system will reboot now!
```
Should the libraries be installed in a venv somewhere?

chvvkumar, 2024-12-09:
Additional troubleshooting:

I tried to manually control the GPIO from a python script (no venv, just directly running the script with python3) by sending the relevant commands via MQTT and it works fine with this script

```
import random
import time
import RPi.GPIO as GPIO
from paho.mqtt import client as mqtt_client

port = 1883
client_id = f'python-mqtt-{random.randint(0, 1000)}'
# Change These
broker = '192.168.1.250'
topic = "rpitest/Relay"
relay_pin = 21

GPIO.setmode(GPIO.BCM)
GPIO.setup(relay_pin, GPIO.OUT)
GPIO.output(relay_pin, GPIO.LOW)

# Set start time
start_time = time.time()

def connect_mqtt():
    def on_connect(client, userdata, flags, rc):
        if rc == 0:
            print("Connected to MQTT Broker!")
        else:
            print("Failed to connect, return code %d\n", rc)
    # Set Connecting Client ID
#    client = mqtt_client.Client(client_id)
    client = mqtt_client.Client(mqtt_client.CallbackAPIVersion.VERSION1, client_id)
    client.on_connect = on_connect
    client.connect(broker, port)
    return client

def subscribe(client: mqtt_client):
    def on_message(client, userdata, msg):
        print(f"Received `{msg.payload.decode()}` from `{msg.topic}` topic")
        if msg.payload.decode() == "ON":
            GPIO.output(relay_pin, GPIO.HIGH)
            print("Relay ON")
        if msg.payload.decode() == "OFF":
            GPIO.output(relay_pin, GPIO.LOW)
            print("Relay OFF")
    client.subscribe(topic)
    client.on_message = on_message

def receive():
    client = connect_mqtt()
    subscribe(client)
    client.loop_forever()

receive()
```

Output that shows pin state change from low to hi and back to low

```
pi@allskypi5:~ $ pinctrl get 21
21: op dl pn | lo // GPIO21 = output

pi@allskypi5:~ $ python3 relay.py
/home/pi/relay.py:28: DeprecationWarning: Callback API version 1 is deprecated, update to latest version
  client = mqtt_client.Client(mqtt_client.CallbackAPIVersion.VERSION1, client_id)
Connected to MQTT Broker!
Received `ON` from `rpitest/Relay` topic
Relay ON

pi@allskypi5:~ $ pinctrl get 21
21: op dh pn | hi // GPIO21 = output

pi@allskypi5:~ $ python3 relay.py
/home/pi/relay.py:28: DeprecationWarning: Callback API version 1 is deprecated, update to latest version
  client = mqtt_client.Client(mqtt_client.CallbackAPIVersion.VERSION1, client_id)
Connected to MQTT Broker!
Received `OFF` from `rpitest/Relay` topic
Relay OFF

pi@allskypi5:~ $ pinctrl get 21
21: op dl pn | lo // GPIO21 = output
```

chvvkumar, 2024-12-09:
I think I might have solved it. Had to activate the venv anduninstall RPi.GPIO python module:

```shell
pi@allskypi5:~/allsky $ systemctl stop allsky

pi@allskypi5:~/allsky $ cd venv/

pi@allskypi5:~/allsky/venv $ . bin/activate

(venv) pi@allskypi5:~/allsky/venv $ pip uninstall RPi.GPIO
Found existing installation: RPi.GPIO 0.7.1
Uninstalling RPi.GPIO-0.7.1:
  Would remove:
    /home/pi/allsky/venv/lib/python3.11/site-packages/RPi.GPIO-0.7.1.dist-info/*
    /home/pi/allsky/venv/lib/python3.11/site-packages/RPi/*
Proceed (Y/n)? y
  Successfully uninstalled RPi.GPIO-0.7.1

(venv) pi@allskypi5:~/allsky/venv $ pip install rpi-lgpio
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: rpi-lgpio in ./lib/python3.11/site-packages (0.6)
Requirement already satisfied: lgpio>=0.1.0.1 in /usr/lib/python3/dist-packages (from rpi-lgpio) (0.2.2.0)

(venv) pi@allskypi5:~/allsky/venv $ deactivate

pi@allskypi5:~/allsky/venv $ systemctl start allsky
```

Result:
```shell
4-12-08T19:11:20.115998-06:00 allskypi5 allskperiodic[8966]: INFO: Loading /home/pi/allsky/config/settings.json
2024-12-08T19:11:20.116265-06:00 allskypi5 allskperiodic[8966]: INFO: ===== Running periodic flow...
2024-12-08T19:11:20.116288-06:00 allskypi5 allskperiodic[8966]: INFO: --------------- Running Module allsky_dewheater ---------------
2024-12-08T19:11:20.116304-06:00 allskypi5 allskperiodic[8966]: INFO: Temperature outside limit temperature 15.67, limit 10, dewPoint -1.44
2024-12-08T19:11:20.116322-06:00 allskypi5 allskperiodic[8966]: INFO: Turning Heater off using pin 21
2024-12-08T19:11:20.116338-06:00 allskypi5 allskperiodic[8966]: INFO: Sensor BME280-I2C read. Temperature 15.67 Humidity 30.9 Relative Humidity 30.768882770967984 Dew Point -1.44 Heat Index -13.12 Pressure 987.9223636286027 Altitude 212.9698522335955
2024-12-08T19:11:20.116356-06:00 allskypi5 allskperiodic[8966]: INFO: --------------- Running Module allsky_pistatus ---------------
2024-12-08T19:11:20.116370-06:00 allskypi5 allskperiodic[8966]: INFO: Will run in 29.318807363510132 seconds
2024-12-08T19:11:20.116385-06:00 allskypi5 allskperiodic[8966]: INFO: --------------- Running Module allsky_publishdata ---------------
2024-12-08T19:11:20.116399-06:00 allskypi5 allskperiodic[8966]: INFO: MQTT - Published to MQTT on channel: Astro/AllSky
2024-12-08T19:11:20.116414-06:00 allskypi5 allskperiodic[8966]: INFO: MQTT - CONNACK received with code Success.
2024-12-08T19:11:20.116428-06:00 allskypi5 allskperiodic[8966]: INFO: ===== periodic flow complete.
2024-12-08T19:11:20.254785-06:00 allskypi5 allskperiodic[8655]: INFO: Sleeping for 30 seconds

```

Alex-developer, 2024-12-09:
Thanks for the info :-)

Was this a clean install of Allsky on a newly installed OS?

chvvkumar, 2024-12-09:
> Thanks for the info :-)
> 
> Was this a clean install of Allsky on a newly installed OS?

The OS was installed new a month ago. 2023.05.01 was what I installed originally.

Steps:

1. Install RPi OS mentioned above and AllSky 2023.05.01 on new Pi5+SSD
2. Install the modules by cloning this repo and running `install.sh`
3. Upgrade AllSky to 2024.12.06 by moving the existing AllSky install to allsky-old (and as such, the venv folder along with it).
4. `git pull` on this repo's local folder in case there were changes and run `install.sh`

Alex-developer, 2024-12-09:
> The OS was installed new a month ago. 2023.05.01 was what I installed originally.
>
> Steps:
>
> 1. Install RPi OS mentioned above and AllSky 2023.05.01 on new Pi5+SSD
> 2. Install the modules by cloning [...]

Thanks, I'll take a look and see if I can reproduce it

chvvkumar, 2024-12-09:
> _EDIT_ Looks like the installer is adding lgpio but NOT removing gpio

Yup, was going to comment just that, both gpio and lgpio seem to exist and removing gpio was enough.

https://github.com/AllskyTeam/allsky-modules/issues/192#issuecomment-2526580327

Alex-developer, 2024-12-09:
> > _EDIT_ Looks like the installer is adding lgpio but NOT removing gpio
> 
> Yup, was going to comment just that, both gpio and lgpio seem to exist and removing gpio was enough.
> 
> [#192 (comment)](https://github.com/AllskyTeam/allsky-modules/issues/192#issuecomment-2526580327)

Thanks, appreciate the investigation - I'll do a fix for it

Alex

chvvkumar, 2024-12-10:
I can confirm that the installer is adding GPIO back. I was installing another module and ran the installer and it added GPIO back in. Removing the module using the above steps made the heater module work again.

Alex-developer, 2024-12-10:
> I can confirm that the installer is adding GPIO back. I was installing another module and ran the installer and it added GPIO back in. Removing the module using the above steps made the heater module work again.

Thanks

I'll have to check what is installing GPIO again ! I might just add a hack to check after installation is complete if its a p5 and remove GPIO

Alex-developer, 2024-12-11:
Change merged that removes rpi.gpio if running on a pi5

closed 2024-12-11 by Alex-developer

chvvkumar, 2024-12-12:
Seems to have done the job, the new image right after installing this still has the overlay variables! Thank you so much!

```shell
Installing allsky_dewheater
===========================
INFO: Installing Python dependencies
INFO: Installing allsky_dewheater module
SUCCESS: Module "allsky_dewheater" installed

INFO: Found the rpi.gpio module so uninstalling it

Found existing installation: RPi.GPIO 0.7.1
Uninstalling RPi.GPIO-0.7.1:
  Successfully uninstalled RPi.GPIO-0.7.1
```