AllskyTeam/allsky-modules #231: Telescope Marker working with Allsky V 2024-12-06

Radinkel, 2025-01-16:
### Discussed in https://github.com/AllskyTeam/allsky-modules/discussions/230

<div type='discussions-op-text'>

<sup>Originally posted by **Radinkel** January 12, 2025</sup>
Hi, 
does the module work with the newest allsky version? 
I'm not sure, but I think I saw the marker in the fallback location in the second to last version. After the update, I configured the module to the best of my knowledge and belief, but the marker is gone, without a trace. As a test, I connected another software to the ASCOM remote server, and I can read all the request and responses in the log there. The gets commands and http://192.168.178.xxx:11111 are identical.
Even fallback position is shown. Image is 1200x1200. My settings:

[image]

Alex-developer, 2025-01-16:
I will have to check fir you, 

Is there anything logged in the allskyperiodic log with the debug level set to 4?

Radinkel, 2025-01-16:
only level 3

Alex-developer, 2025-01-16:
> only level 3

Can you set the debug level to 4 and see if there is anything logged.

If not I will setup my scope at the weekend and see if I can reproduce this

Radinkel, 2025-01-16:
Good news: Ascom remote is logging requests from the allsky, and gets a response without error. But still no marker, even at fallbak position

Radinkel, 2025-01-16:
Set to level 4. Sorry, but I need more detailled instructions: How to get the log file? All *.log are some days old.

Alex-developer, 2025-01-16:
> Set to level 4. Sorry, but I need more detailled instructions: How to get the log file? All *.log are some days old.

No problem, run the following

`tail -f /var/log/allskyperiodic.log`

Radinkel, 2025-01-16:
I love linux. Who can remember all the commands? In Windows there is always something to click on, unfortunately the button is always somewhere else...

Radinkel, 2025-01-16:
Does this help?
I see other modules, but no telescope marker. Maybe module is not loaded.
I think before updating to V24-12-06 I saw the red circle on the fallback position.

admin@allsky:~ $ tail -f /var/log/allskyperiodic.log
2025-01-16T23:35:29.580149+01:00 allsky allskperiodic[3641]: INFO: Turning Heate                  r off using pin 23
2025-01-16T23:35:29.580198+01:00 allsky allskperiodic[3641]: INFO: Sensor DHT22                   read. Temperature 23.5 Humidity 32.4 Relative Humidity None Dew Point 6.04 Heat                   Index -8.29 Pressure None Altitude None
2025-01-16T23:35:29.580249+01:00 allsky allskperiodic[3641]: INFO: -------------                  -- Running Module allsky_pistatus ---------------
2025-01-16T23:35:29.580289+01:00 allsky allskperiodic[3641]: INFO: PI Status Dat                  a Written
2025-01-16T23:35:29.580424+01:00 allsky allskperiodic[3641]: INFO: -------------                  -- Running Module allsky_weatherunderground ---------------
2025-01-16T23:35:29.580471+01:00 allsky allskperiodic[3641]: ERROR:
2025-01-16T23:35:29.580526+01:00 allsky allskperiodic[3641]: INFO: -------------                  -- Running Module allsky_spaceweather ---------------
2025-01-16T23:35:29.580568+01:00 allsky allskperiodic[3641]: INFO: Last run 123.                  3262403011322 seconds ago. Running every 300 seconds
2025-01-16T23:35:29.580619+01:00 allsky allskperiodic[3641]: INFO: ===== periodi                  c flow complete.
2025-01-16T23:35:29.819583+01:00 allsky allskperiodic[713]: INFO: Sleeping for 6                  0 seconds
2025-01-16T23:36:31.105622+01:00 allsky allskperiodic[3670]: INFO: Loading /home/admin/al         lsky/config/settings.json
2025-01-16T23:36:31.106400+01:00 allsky allskperiodic[3670]: INFO: ===== Running periodic          flow...
2025-01-16T23:36:31.106518+01:00 allsky allskperiodic[3670]: INFO: --------------- Runnin         g Module allsky_dewheater ---------------
2025-01-16T23:36:31.106565+01:00 allsky allskperiodic[3670]: INFO: Temperature outside li         mit temperature 23.7, limit 10, dewPoint 6.26
2025-01-16T23:36:31.106665+01:00 allsky allskperiodic[3670]: INFO: Turning Heater off usi         ng pin 23
2025-01-16T23:36:31.106730+01:00 allsky allskperiodic[3670]: INFO: Sensor DHT22 read. Tem         perature 23.7 Humidity 32.5 Relative Humidity None Dew Point 6.26 Heat Index -8.17 Pressu         re None Altitude None
2025-01-16T23:36:31.106791+01:00 allsky allskperiodic[3670]: INFO: --------------- Runnin         g Module allsky_pistatus ---------------
2025-01-16T23:36:31.106837+01:00 allsky allskperiodic[3670]: INFO: PI Status Data Written
2025-01-16T23:36:31.106892+01:00 allsky allskperiodic[3670]: INFO: --------------- Runnin         g Module allsky_weatherunderground ---------------
2025-01-16T23:36:31.106934+01:00 allsky allskperiodic[3670]: INFO: Last run 61.5291175842         28516 seconds ago. Running every 120 seconds
2025-01-16T23:36:31.106985+01:00 allsky allskperiodic[3670]: INFO: --------------- Runnin         g Module allsky_spaceweather ---------------
2025-01-16T23:36:31.107314+01:00 allsky allskperiodic[3670]: INFO: Last run 184.852155923         84338 seconds ago. Running every 300 seconds
2025-01-16T23:36:31.107422+01:00 allsky allskperiodic[3670]: INFO: ===== periodic flow co         mplete.
2025-01-16T23:36:31.357741+01:00 allsky allskperiodic[713]: INFO: Sleeping for 60 seconds

Radinkel, 2025-01-16:
Lots of errors:
1. I made the image square, as requested (is this really a requirement?), now several layout elements no longer fit
2. The FTP upload is not working for some reason, I suspect the login takes too long and the temporary files are cluttering up the NAS. I will delete them.

admin@allsky:~ $ tail -f /var/log/allsky.log
2025-01-16T23:45:39.554286+01:00 allsky allsky[4762]:    *** upload.sh: ERROR, R                                                                    ET=2:
2025-01-16T23:45:39.554468+01:00 allsky allsky[4762]: FILE_TO_UPLOAD='/home/admi                                                                    n/allsky/tmp/image-20250116234521.jpg'
2025-01-16T23:45:39.554620+01:00 allsky allsky[4762]: REMOTE_HOST='ehcj2i7etqqin                                                                    lyu.myfritz.net'
2025-01-16T23:45:39.554743+01:00 allsky allsky[4762]: DIRECTORY='Spica/allsky//'
2025-01-16T23:45:39.554903+01:00 allsky allsky[4762]: TEMP_NAME='SaveImage-serve                                                                    r-18759'
2025-01-16T23:45:39.555070+01:00 allsky allsky[4762]: DESTINATION_NAME='image.jp                                                                    g'
2025-01-16T23:45:39.555194+01:00 allsky allsky[4762]: put: /home/admin/allsky/tm                                                                    p/image-20250116234521.jpg: Access failed: 553 SaveImage-server-18759: No space                                                                     left on device.
2025-01-16T23:45:39.555341+01:00 allsky allsky[4762]: put of /home/admin/allsky/                                                                    tmp/image-20250116234521.jpg to SaveImage-server-18759 failed!  Trying again...
2025-01-16T23:45:39.555502+01:00 allsky allsky[4762]: put: /home/admin/allsky/tm                                                                    p/image-20250116234521.jpg: Access failed: 553 SaveImage-server-18759: No space                                                                     left on device.
2025-01-16T23:45:39.555617+01:00 allsky allsky[4762]: 2nd put failed again, quit                                                                    ting
2025-01-16T23:47:24.010659+01:00 allsky allsky[886]: Executing sunwait poll exit angle -6.0000 49.496077N 11.2724                                   23E > /dev/null
2025-01-16T23:47:24.017330+01:00 allsky allsky[886]: -----
2025-01-16T23:47:24.017631+01:00 allsky allsky[886]: STARTING EXPOSURE at: 2025-01-16 23:47:24   @ 52.56 ms (0.05                                    sec)
2025-01-16T23:47:24.102393+01:00 allsky allsky[886]:  > Running: rpicam-still --thumb none --output '/home/admin/                                   allsky/tmp/image-20250116234724.jpg' --timeout 1 --nopreview --width 4056 --height 3040 --shutter 52556 --analogg                                   ain 1 --awbgains 3,2 --rotation 180 --saturation 0.8 --quality 100
2025-01-16T23:47:26.237734+01:00 allsky allsky[886]:   > Got:    shutter_us: 52.56 ms (0.05 sec), gain: 1.000, me                                   an: 0.141, target mean: 0.150, diff (target - mean): 0.009
2025-01-16T23:47:26.237929+01:00 allsky allsky[886]:   > index: 0, meanHistory[]=0.142 exposureLevelHistory[]=-15                                   3, newNean=0.142
2025-01-16T23:47:26.237990+01:00 allsky allsky[886]:   > index: 1, meanHistory[]=0.141 exposureLevelHistory[]=-15                                   3, newNean=0.424
2025-01-16T23:47:26.238048+01:00 allsky allsky[886]:   > index: 2, meanHistory[]=0.141 exposureLevelHistory[]=-15                                   3, newNean=0.846
2025-01-16T23:47:26.238099+01:00 allsky allsky[886]:   > New mean target: 0.141, mean_forecast: 0.141, mean_diff                                    (newMean - target mean): 0.009, idx=2, idxN1=1
2025-01-16T23:47:26.238149+01:00 allsky allsky[886]:   > ExposureChange clipped to 3 (diff from last change: 0)
2025-01-16T23:47:26.238205+01:00 allsky allsky[886]:   > ++++++++++ Prior image mean good - no changes needed, me                                   an=0.141, target mean=0.150 threshold=0.010
2025-01-16T23:47:26.238247+01:00 allsky allsky[886]:   > Saving NIGHT image 'image-20250116234724.jpg'
2025-01-16T23:47:26.239316+01:00 allsky allsky[886]:   > Sleeping: 119.9 sec
2025-01-16T23:47:26.244951+01:00 allsky allsky[4882]: saveImage.sh NIGHT /home/admin/allsky/tmp/image-20250116234                                   724.jpg DATE=20250116 TIME=234724 AUTOEXPOSURE=1 sAUTOEXPOSURE=(auto) EXPOSURE_US=52556 sEXPOSURE=52.56 ms (0.05                                    sec) AUTOGAIN=1 sAUTOGAIN=(auto) GAIN=1 AUTOWB=0 sAUTOAWB= WBR=3 WBB=2 MEAN=0.140888 BIN=1 FLIP=none BIT_DEPTH=0                                    DARKFRAME=0 eOVERLAY=1
2025-01-16T23:47:27.253091+01:00 allsky allsky[4882]: saveImage.sh Cropping '/home/admin/allsky/tmp/image-2025011                                   6234724.jpg' to 3040x3040.
2025-01-16T23:47:30.660990+01:00 allsky allsky[4926]: INFO: Loading /home/admin/allsky/config/settings.json
2025-01-16T23:47:30.661287+01:00 allsky allsky[4926]: INFO: ===== Running night flow...
2025-01-16T23:47:30.661371+01:00 allsky allsky[4926]: INFO: --------------- Running Module allsky_loadimage -----                                   ----------
[... cut ...]
2025-01-16T23:47:41.702060+01:00 allsky allsky[4999]: put: /home/admin/allsky/tmp/image-20250116234724.jpg: Acces                                   s failed: 553 SaveImage-server-8230: No space left on device.
2025-01-16T23:47:41.702211+01:00 allsky allsky[4999]: put of /home/admin/allsky/tmp/image-20250116234724.jpg to S                                   aveImage-server-8230 failed!  Trying again...
2025-01-16T23:47:41.702352+01:00 allsky allsky[4999]: put: /home/admin/allsky/tmp/image-20250116234724.jpg: Acces                                   s failed: 553 SaveImage-server-8230: No space left on device.
2025-01-16T23:47:41.702466+01:00 allsky allsky[4999]: 2nd put failed again, quitting

Alex-developer, 2025-01-16:
This looks like the issue.

`Module allsky_telescopemarker.py failed on line 209 - could not convert string to float: '49.496077N'`

The lat and lon is expected to be decimal without the 'N', 'S', 'E' or 'W'

Make sure your lat and lon are just numbers with out the cardinal letter

Radinkel, 2025-01-16:
East means -11....?

Alex-developer, 2025-01-16:
> East means -11....?

Correct

Radinkel, 2025-01-16:
OK, I had left the fields empty at first because the allsky settings are used by default and they had to be entered with N, E etc. Perhaps this has changed in the version from 24-12-06? That would be an explanation?

Radinkel, 2025-01-16:
Now:
2025-01-17T00:11:58.474028+01:00 allsky allsky[7633]: INFO: --------------- Running Module allsky_telescopemarker ---------------
2025-01-17T00:11:58.474068+01:00 allsky allsky[7633]: INFO: --------------- Running Module allsky_overlay ---------------

No error?!

Radinkel, 2025-01-16:
The marker has appeared!!!!!!!!!!!!!!!!!!!!
Thank you!!!!!!!!!!!!!!!!!

Radinkel, 2025-01-16:
I'll have to test on the telescope tomorrow whether the data from the Ascom Remote Server is read. I'm at home now ;)

Does the image have to be square?

Radinkel, 2025-01-17:
Dangerous:
here "-" means obviously "W"

[image]

Alex-developer, 2025-01-17:
I didn't write the module so not sure if the image needs @frankhirsch can you comment on this and the lat/lon?

Thanks

Radinkel, 2025-01-17:
No, I meant that since there are obviously different formatting options within Allsky, this should be indicated under each input field. Unfortunately, in your case, Alex, it is missing.

Radinkel, 2025-01-17:
Marker is working, even comunication to ASCOM Remote Server.
Thank you Alex!

frankhirsch, 2025-01-17:
Happy you finally got it working! I will check to be more precise in the comments of the configuration, but I think keeping the Location is useful if camera and telescope are on different locations.

Radinkel, 2025-01-17:
@frankhirsch: YOU made it right: You did a example under the input box....

Radinkel, 2025-02-22:
Hi,
after it worked once, it no longer works since the ASCOM remote server is running on a different computer. I copied the settings 1:1. Nothing changed on the Pi. The remote server is up and drivers are connected. But there are no requests and certainly no replies. I assume it's because of super-windows, I've already turned off the firewall, but that's not the problem.

There is no entry in the allsky.log under "Running Module allsky_telescopemarker". The marker stays in the default position the whole time.

Any ideas?

frankhirsch, 2025-02-23:
If the telescope marker is not moving (showing in default position) it‘s an indicator the connection to ASCOM Server did not succeed. No issue to calculate the marker position occurred… confirmed by the fact no traffic occurs in ASCOM Server itself (req/res).

Did you change in ASCOM Server the listening IP to the LAN IP (not localhost only)? Sounds like this should be your issue.

Radinkel, 2025-02-23:
Hi Frank, 
here settings of the plugin:

[image]

here console ascom remote server: server up, driver connected, no req/res

[image]

Marker on fallback position:

[image]

Telescope pointing to the sky (Az 185 / Alt 50).

[image]

frankhirsch, 2025-02-23:
In ASCOM server click „setup“ in the bottom right corner. In the second(?) tab change the listener (server) IP from 127.0.0.1 to your LAN IP. Otherwise ASCOM server will not respond to network requests!

[image]

frankhirsch, 2025-02-23:
127.0.0.1 is localhost (internal loopback device). That‘s a very common thing for network interfaces across all sorts of operating systems. In case you do not know your local IP address I recommend to check your network configuration.

Also recommended: if your computer (running ASCOM server) does get a different local IP address in your network every now and then I highly recommend to configure a static IP or make a lease reservation in your router/DHCP server. Otherwise you will have the same issue again if the IP changes after some time!

Radinkel, 2025-02-23:
Thank you for your reply. This is what happens when you don't understand exactly what you're doing. 
I have set both the remote server and plugin to an IP in my network. Windows now denies the remote server the right to listen on this IP, even though I have admin rights. I think we are on the right track. I just have to convince Windows of this and that gets harder with every new version!

Radinkel, 2025-02-23:
OK, I think I understand now: I have to enter the IP of the computer on which the remote server is running. Since I changed the computer, the IP is also different...

frankhirsch, 2025-02-23:
Agree… ☺️

Now go to security settings & firewall in your Windows OS and allow incoming TCP Port 11111 for your network profile (or all profiles)

[image]

Just a side note… I am using ASCOM Remote on an Intel NUC mounted directly to the telescope - feeding all my gear (camera, EAF/EFW…) remotely to my imaging server. In case you have such a setup, you should consider disabling Windows Defender Live Antivirus scan - this will significantly impact the performance while shooting!

PS: You are from Herrenberg (Brunnenbau)?

Radinkel, 2025-02-23:
Thank you! It works if it is explained to you in detail ;)
Thanks for the hint! I switched to an eagle6 (also NUC inside), and its working quite slow. Maybe this will speed it up!
I have two rigs. Maybe its possible that the plugin can mark one position in yellow and the other in red, for example.

P.S. I worked almost 20 years for OCHS Brunnenbau, Nürnberg. Jetzt habe ich ein eigenes Ingenieurbüro für Brunnenbau.

frankhirsch, 2025-02-23:
That‘s feasible… but not out of the box 😜 You may duplicate the plugin in AllSky (changing manually the name and so on…). In the plugin itself you can change the marker color (RGB) per instance.

There are thoughts @Alex-developer for AllSky to use multiple instances of one plugin in the future, but this is not on the roadmap as far as I know.

As for the performance… I decided to install the free version of AVG (antivirus) on my NUC since disabling Windows 11 Defender Live Scan does not seem possible anymore (once disabled it get’s reactivated every 24 hours or so). Installing another Antivrus will disable Windows Defender. Since I do not use my NUC to browse the internet I feel still pretty safe!

PS: Dann kennst Du das wohl https://www.gis.uni-stuttgart.de/BFO/ ?

frankhirsch, 2025-02-23:
> Does the image have to be square?

Yes, the protection presumes the image is cropped to square as the margin is considered unique for height & width in the projection model.

jcauthen78, 2025-02-23:
FYI - I'm currently in work on getting this working on a rectangular image, with multiple tabs for the module for various settings. Here are some screenshots of where i'm at with it - still working on the projection method config. 

[image]

[image]

[image]

[image]

[image]

jcauthen78, 2025-02-23:
Here's the view on a rectangular image with the alignment crosshairs active (had to brighten up the image really far to see the extent of the actual imaging ring, was a lot bigger than I originally setup). The color of the ring allows for an alpha/transparency option to make the marker a little less 'bold' once it's setup where ya want it. 
  
[image]

For the networking side, I had to set a static IP on my telescope imaging rig for the connection to stay reliable.

frankhirsch, 2025-02-23:
@jcauthen78 Jim, that‘s really cool!

jcauthen78, 2025-02-23:
appreciate it @frankhirsch - Tonight is 100% clouded & rainy, so can't test the projection adjustments yet - Unsure if I implemented them correctly or if they're being correctly applied.... Tomorrow night is looking ok for the first few hours of darkness to slew the telescope around to some of the brighter stars/planets.  

For you and @Radinkel (and anyone else following) - here's the script as it sits right now so you can take a look at it and see what you think. I'm not good with GitHub, so not sure how to save/share/push/pull and all that stuff. But hopefully this python (renamed to .txt because can't share a .py file in this manner) will work for a copy & paste. It's working without issue on my end (save for the projection adjustments), so hopefully that translates. 

[allsky_telescopemarker.txt](https://github.com/user-attachments/files/18932975/allsky_telescopemarker.txt)

Radinkel, 2025-02-23:
That looks really nice! I'm very interested. I would also like to display additional data from my weather station (including SQM). We have clear skies here in germany ;)

jcauthen78, 2025-02-23:
> That looks really nice! I'm very interested. I would also like to display additional data from my weather station (including SQM). We have clear skies here in germany ;)

Not to get off topic, but, If you can have it output its readings to a basic JSON file (~/allsky/config/overlay/extra/weather.json (or whatever you want to name it)), AllSky can import the values into the overlay editor without too much struggle. It's a powerful little tool to be able to use the extra files for more overlay data. 

[image]

Feel free to shoot a message over if you have questions about it and I'll show ya some examples of what generates them for me.

Radinkel, 2025-02-23:
I use AAG-cloudwatcher from lunaticoastro. Up to now I use a txt but export to a json should possible too.

frankhirsch, 2025-02-23:
﻿
or… you can use the ASCOM Remote Server as done for the telescope marker 😉
That’s JSON out of the box…

Radinkel, 2025-02-23:
@frankhirsch: that's a good idea. Together we are strong. I started with astrophotography almost 40 years ago. It was unimaginable what was possible today!!

frankhirsch, 2025-02-23:
I only started shortly before ASIAIR v1 was launched around 2007 - having no friends doing this hobby I was doing trial and error from scratch with not even a clue of astronomy 🥹 Platesolving was probably one of the techniques allowing people like me to do the hobby without the usual “beginner frustration…”

But today I also know people which did analogue deep sky photography while guiding manually! Hard to imagine today with all these helpers…

frankhirsch, 2025-02-23:
> [@frankhirsch](https://github.com/frankhirsch): that's a good idea. Together we are strong. I started with astrophotography almost 40 years ago. It was unimaginable what was possible today!!

That should do it: https://github.com/AllskyTeam/allsky-modules/tree/master/allsky_jsonimport

Radinkel, 2025-02-23:
Thank you!
But what's now going on? 

[image]

Radinkel, 2025-04-19:
I don't know, whats going on, but it's not working anymore. 
I run `tail -f /var/log/allskyperiodic.log`, other modules are working, but module telescope marker is not mentioned in the list (Debug level 4). So it's not running. What else can I do, then enable the module and save....
Ralf

Alex-developer, 2025-04-20:
Is the module enabled? If you look in the module manager ensure that the checkbox is checked for the module.

If it is enabled then perhaps @frankhirsch can help

Radinkel, 2025-04-20:
Yes, enabled. Firewall disabled (because is't not reachable from internet)

[image]

What does this mean: Debug Info: telescopemarker false
@frankhirsch please assist!

frankhirsch, 2025-04-20:
I will try to check tomorrow. But I presume the screenshot would be read „Loaded false“ - the reason it is not loaded should be in the default „allsky.log“. Maybe there are missing Python dependcies?

Radinkel, 2025-04-20:
allsky.log:
ERROR: Module allsky_telescopemarker.py failed on line 209 - Expecting value: line 1 column 1 (char 0)
Does this help further?

frankhirsch, 2025-04-20:
Kurz zum Abgleich - welchen Build von AllSky und welches Python verwendest Du?

Radinkel, 2025-04-20:
Allsky Version: v2024.12.06
Python: ähm, ist bei mir eine Würgeschlage...

frankhirsch, 2025-04-21:
I am a bit lost since line #209 in the script is a comment block - not sure what could be wrong here. Maybe @Alex-developer can help to debug & identify…?

1. Reinstall the module from the repository (this will also install any missing dependencies in the python environment)
2. Check for potential issues in the configuration (has the index of the telescope changed in your ASCOM server?)

frankhirsch, 2025-04-21:
> Allsky Version: v2024.12.06 Python: ähm, ist bei mir eine Würgeschlage...

Running the following command will give you the Python version in use

allsky@allsky:~/allsky $ /home/allsky/allsky/venv/bin/python
Python 3.11.2 (main, May  2 2024, 11:59:08) [GCC 12.2.0] on linux

frankhirsch, 2025-04-21:
Currently in the update process to check if there are issues in more recent versions...

frankhirsch, 2025-04-21:
Ok, I updated to the most recent OS patches and the current Allsky version v2024.12.06_02. There does not seem to be a generic issue...

Can you please post the content of **~/allsky/config/modules/postprocessing_night.json** to check if you have potential errors in your's? 

There should be a section for "telescopemarker" looking like this...
```
[...]
    "telescopemarker": {
        "module": "allsky_telescopemarker.py",
        "metadata": {
            "name": "Telescope postion marker",
            "description": "Mark the current telescope postion retreived from ASCOM Remote Server in the image",
            "module": "allsky_telescopemarker",
            "version": "v0.1",
            "events": {
                "0": "night",
                "1": "day"
            },
            "experimental": true,
            "arguments": {
                "observer_lat": "48.58",
                "observer_lon": "8.0",
                "observer_height": "130",
                "image_lat": "48.58",
                "image_lon": "8.0",
                "image_height": "132",
                "image_flip": "Horizontal",
                "camera_azimuth": "200",
                "margin": "60",
                "telescope_marker_radius": "30",
                "telescope_marker_width": "5",
                "telescope_marker_color": "(0,0,255)",
                "telescope_server": "http://192.168.178.193:11111",
                "telescope_alt": "/api/v1/telescope/0/altitude",
                "telescope_az": "/api/v1/telescope/0/azimuth",
                "telescope_default": "(0.0,0.0)"
            },
[...]
```

Alex-developer, 2025-04-22:
> I am a bit lost since line 209 in the script is a comment block - not sure what could be wrong here. Maybe [@Alex-developer](https://github.com/Alex-developer) can help to debug & identify…?
> 
> 1. Reinstall the module from the repository (this will also install any missing dependencies in the python environment)
> 2. Check for potential issues in the configuration (has the index of the telescope changed in your ASCOM server?)

Frank

Thats a 'glitch' in the error handling. Line 209 refers to flow-runner.py not the module :-(

The best option is to remove the try catch in flow-runner so that you get the true line in the module causing the error. In the next major release I have change this so that all modules report the error on the correct line

Radinkel, 2025-04-22:
#

> Can you please post the content of ~/allsky/config/modules/postprocessing_night.json to check if you have potential errors in your's? Of course - arguments will be your personal configuration!

Mine is much shorter:

[image]

I've made a strange new observation: When the Ascom Remote Server at 192.168.177.30 (control PC) is off, the marker is displayed at the fallback position. When the server is on, the marker disappears, but the server doesn't record any requests.

[image]

frankhirsch, 2025-04-22:
> Frank
>
> Thats a 'glitch' in the error handling. Line 209 refers to flow-runner.py not the module :-(
>
> The best option is to remove the try catch in flow-runner so that you get the true line in the [...]

Ok, sounds good @Alex-developer 

In the specific case this would still not help since it seems something seems potentially wrong in the handling of input values. Running Python ast.literal_eval against a string seems a bit dangerous if the input is not validated… any recommendation/idea how we can/should assure users will use the right format when configuring modules?

Radinkel, 2025-04-22:
@Alex-developer 
What is the exact grep command to view the entire Allsky.log?

[9 more replies]