AllskyTeam/allsky-modules #233: allsky_temp, too low humidity with BME280

Alex-developer, 2025-01-17:
### Discussed in https://github.com/AllskyTeam/allsky-modules/discussions/232

<div type='discussions-op-text'>

<sup>Originally posted by **chvvkumar** January 17, 2025</sup>
### Scenario
I have a desiccant bag in my AllSky enclosure. This keeps the humidity quite low in addition to the pressure compensating vents I installed.

### Issue
From what I can tell, the allsky_temp module considers any relative humidity value <1% as invalid. But my BME280 is reporting a relative humidity of 0.645%. This causes the module to fail to save this extra variable.

### Relevant logs
```shell
pi@allskypi5:~ $ tail -f /var/log/allskyperiodic.log
.
.
25-01-17T14:55:05.385625-06:00 allskypi5 allskperiodic[5843]: INFO: --------------- Running Module allsky_temp ---------------
2025-01-17T14:55:05.385642-06:00 allskypi5 allskperiodic[5843]: INFO: Reading sensor 1, BME280-I2C
2025-01-17T14:55:05.385657-06:00 allskypi5 allskperiodic[5843]: ERROR: Module allsky_temp.py failed on line 209 - Incorrect value for humidity: "0.6450005825294641". Correct range 1-100.
.
.
```

### Troubleshooting steps so far
1. Checked I2C cabling which seems to be correct also corroborated by the fact that the other BME280 readings report sane values
2. Run a python script outside of AllSky to read the sensor values and these readings match AllSky readings so far

```python
import smbus2
import bme280

port = 1
address = 0x76
bus = smbus2.SMBus(port)

calibration_params = bme280.load_calibration_params(bus, address)

# the sample method will take a single reading and return a
# compensated_reading object
data = bme280.sample(bus, address, calibration_params)

# there is a handy string representation too
print(data)
```
Output:

```shell
(venv) pi@allskypi5:~/bme280 $ python bm.py
compensated_reading(id=b9823947-0f68-4a83-8f5d-09c38c873e47, timestamp=2025-01-17 21:02:40.408275UTC, temp=20.367 °C, pressure=983.62 hPa, humidity=0.46 % rH)
```

As can be seen from the values over this past week, Rh has been falling over time so I think the sensor itself is working:

[image]
</div>

Alex-developer, 2025-02-02:
I have found the issue with this and will release a fix in the next release. The sensor is 100% ok its one of the internal libraries that calculates the dew point, its only allowing a value between 1 and 100.

drfritz142, 2025-08-25:
I have the exact same trouble... has it been fixed yet? in the meantime... is there a quick-n-dirty hack? just bump the humidity to 1% if it is smaller than 1? I'd do it, but which python script do I need to edit? From the error information (line 209 - in my case by now: 212) I cannot conclude which file I need to modify. Is it  the one located in /opt/allsky/modules/....?