AllskyTeam/allsky-modules #252: allsky_ina3221 requirements.txt file name error

twostar01, 2025-03-05:
**Describe the bug**
In the allsky_ina3221 module, the requirements.txt file name has a comma instead of period. This prevents successful installation of the required dependencies and therefore use of the module. 

**To Reproduce**
Try installing module ina3221 via Allsky module installer.

Alex-developer, 2025-03-09:
Apologies for the late reply

I have merged the two changes, thanks for those.

I am also in the process of a major upgrades to the modules and have renamed the ina3221 module to allsky_power as it supports the ina219 now as well. This update is a little way off but I will add your changes to the new allsky_power module

The new module is at [allsky_power
](https://github.com/AllskyTeam/allsky/blob/master/scripts/modules/allsky_export.py) BUT this cannot be used with the current version of allsky as it requires a lot of other changes.

Thanks

Alex

twostar01, 2025-03-10:
No need to apologize, figured it would take a bit to get merged and there's no rush. I've already updated my system with the modifications so I know they work.

Good to know about the new module. I'm running my system off of solar power with battery storage so I'm working out how to track the overall power usage with the ina3221. Would you like me to keep future updates along these lines in the power module? I'm going to add rough battery status (charging, discharging, low battery, etc) to my system and was thinking of either writing a custom module for it or integrating it into the ina3221 module. Now that you're standing up a power module it might make more sense to go there.

[image]Alex-developer left a comment (AllskyTeam/allsky-modules#252) <https://github.com/AllskyTeam/allsky-modules/issues/252#issuecomment-2709119271>

Alex-developer, 2025-03-10:
> Good to know about the new module. I'm running my system off of solar power with battery storage so I'm working out how to track the overall power usage with the ina3221. Would you like me to keep [...]

That would be great. If you want to build the code into the 3221 module for now when the power module is ready I can let you know.

Just to give some idea of the changes

1) Improved UI, fields are only visible when they are needed
2) Lots of new 'plugin's to select stuff, like the i2c example in the video. It detects the i2c devices connected and shows you what they are, no more guessing i2c addresses :-)
3) Module testing. Allows you to test the module settings
4) Shows the variables that would be created for the overlays
5) The variable manager in the overlay editor has been replaced with a single list of all available variables no more having to add them from one tab to the other, see screenshot below

https://github.com/user-attachments/assets/4fe81a13-56a8-4c35-ad6c-b9ede1601050

[image]