AllskyTeam/allsky-modules #279: allsky_fans error

drakula1981, 2025-08-31:
**Describe the bug**
On a brand new install for my allsky camera on my PI5 with the last OS available, I want to control fans  in PWM mode with the module allsky_fans.
But I struggle with the following error : 
"Module allsky_fans.py failed on line 212 - Need to add 'dtoverlay=pwm-2chan' to /boot/config.txt and reboot"
Of course, I added the following line dtoverlay=pwm-2chan,pin=18,func=4,pin2=19,func2=4 at he end of the /boot/firmware/config.txt file and it's still doesn't works
Any Idea about what i've could missed or did wrong?

**Allsky Information:**
 - OS: Raspberrypi OS bookworm lite 64 bits 
 - Allsky Version: v2024.12.06_06
 - Pi model: Raspberry Pi 5 Model B Rev 1.0, 4 GB
 - Camera: Zwo ASI678MC

**Additional context**
fan connected to PIN18

Alex-developer, 2025-10-06:
It's probably best to wait for the next release which has a lot of fixes for PWM fan control. Its very different between the Pi4 and 5 so I have made a lot of changes to support both.

drakula1981, 2026-10-27:
I've installed the v2026.10.01 allsky software and I still have issues with allsky_fans module. (either v1.0.3 or v1.0.4)
When I activate the module I have the following loop error :
`ERROR in /home/pi/allsky/config/myFiles/modules/allsky_fans.py: Fan 1 - Module _use_bool_fan_control - 817 - 500 Server Error: INTERNAL SERVER ERROR for url: http://localhost:8090/gpio/digital &nbsp; (6 occurrences, last on October 27,  6:39:20 pm CET)`
I didn't activated pwm yet, just activated the "activate fan" in the fan 1 tab, selected the AS_CPUTEMP in the variable selector set 60°C in tempareture limit, read every 60s and set the GPIO18 (pin 12) as the output pin.
I'am avalaible to give further informations

Alex-developer, 2026-10-27:
> 3\. I've installed the v2026.10.01 allsky software and I still have issues with allsky_fans module. (either v1.0.3 or v1.0.4)
>     When I activate the module I have the following loop error :
>     `ERROR in /home/pi/allsky/config/myFiles/modules/allsky_fans.py: Fan 1 - Module _use_bool_fan_control - 817 - 500 Server Error: INTERNAL SERVER ERROR for url: http://localhost:8090/gpio/digital &nbsp; (6 occurrences, last on October 27,  6:39:20 pm CET)`
>     I didn't activated pwm yet, just activated the "activate fan" in the fan 1 tab, selected the AS_CPUTEMP in the variable selector set 60°C in tempareture limit, read every 60s and set the GPIO18 (pin 12) as the output pin.
>     I'am avalaible to give further informations

Sorry  Iseem to have missed this.

Can you post the support info?

Also We have w new release coming out in the next week or so that adds more logging to the server process that manages the GPIO as currently its hard to see whats going on

Thanks

Alex

drakula1981, 2026-11-03:
Hello,
Sorry for the late reply.
I’m not sure exactly what you need; I went to the 'Getting Support' section and generated a package, but I can’t upload it because the .zip file is 28 MB.
I have installed version v2026.10.01_01, and when I go to 'System' -> 'Logs' and select the 'Allsky Server log', I see the following errors repeating:

```
026-11-03T21:45:03+0100 gunicorn[844]: 2026-11-03 21:45:03 xGpioHandleRequest: Invalid argument
2026-11-03T21:45:03+0100 gunicorn[844]: 2026-11-03 21:45:03,526 ERROR modules.gpio /home/pi/allsky/server/modules/gpio/__init__.py:545: Failed to set digital pin
2026-11-03T21:45:03+0100 gunicorn[844]: Traceback (most recent call last):
2026-11-03T21:45:03+0100 gunicorn[844]:   File "/home/pi/allsky/server/modules/gpio/__init__.py", line 529, in set_digital
2026-11-03T21:45:03+0100 gunicorn[844]:     _claim_output(pin_num, 1 if state else 0)
2026-11-03T21:45:03+0100 gunicorn[844]:   File "/home/pi/allsky/server/modules/gpio/__init__.py", line 116, in _claim_output
2026-11-03T21:45:03+0100 gunicorn[844]:     get_gpio_backend().claim_output(pin_num, level=level)
2026-11-03T21:45:03+0100 gunicorn[844]:   File "/home/pi/allsky/server/modules/gpio/backends/lgpio_backend.py", line 158, in claim_output
2026-11-03T21:45:03+0100 gunicorn[844]:     self.lgpio.gpio_claim_output(self._handle(), pin_num, level=level)
2026-11-03T21:45:03+0100 gunicorn[844]:   File "/usr/lib/python3/dist-packages/lgpio.py", line 781, in gpio_claim_output
2026-11-03T21:45:03+0100 gunicorn[844]:     return _u2i(_lgpio._gpio_claim_output(handle&0xffff, lFlags, gpio, level))
2026-11-03T21:45:03+0100 gunicorn[844]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2026-11-03T21:45:03+0100 gunicorn[844]:   File "/usr/lib/python3/dist-packages/lgpio.py", line 458, in _u2i
2026-11-03T21:45:03+0100 gunicorn[844]:     raise error(error_text(v))
2026-11-03T21:45:03+0100 gunicorn[844]: lgpio.error: 'unexpected error'
2026-11-03T21:45:03+0100 gunicorn[844]: 2026-11-03 21:45:03,527 INFO app /home/pi/allsky/server/app.py:162: HTTP POST /gpio/digital -> 500
```
Hope this Helps

EricClaeys, 2026-11-04:
@drakula1981,
Would you please upload `~/allsky/config/logs/support.log` ?
This release is supposed to make sure the support log isn't too large.

@Alex-developer will address your other issue.

Alex-developer, 2026-11-04:
> 5\. Hello,
>     Sorry for the late reply.
>     I’m not sure exactly what you need; I went to the 'Getting Support' section and generated a package, but I can’t upload it because the .zip file is 28 MB.
>     I have installed version v2026.10.01_01, and when I go to 'System' -> 'Logs' and select the 'Allsky Server log', I see the following errors repeating:
>     ```
>     026-11-03T21:45:03+0100 gunicorn[844]: 2026-11-03 21:45:03 xGpioHandleRequest: Invalid argument
>     2026-11-03T21:45:03+0100 gunicorn[844]: 2026-11-03 21:45:03,526 ERROR modules.gpio /home/pi/allsky/server/modules/gpio/__init__.py:545: Failed to set digital pin
>     2026-11-03T21:45:03+0100 gunicorn[844]: Traceback (most recent call last):
>     2026-11-03T21:45:03+0100 gunicorn[844]:   File "/home/pi/allsky/server/modules/gpio/__init__.py", line 529, in set_digital
>     2026-11-03T21:45:03+0100 gunicorn[844]:     _claim_output(pin_num, 1 if state else 0)
>     2026-11-03T21:45:03+0100 gunicorn[844]:   File "/home/pi/allsky/server/modules/gpio/__init__.py", line 116, in _claim_output
>     2026-11-03T21:45:03+0100 gunicorn[844]:     get_gpio_backend().claim_output(pin_num, level=level)
>     2026-11-03T21:45:03+0100 gunicorn[844]:   File "/home/pi/allsky/server/modules/gpio/backends/lgpio_backend.py", line 158, in claim_output
>     2026-11-03T21:45:03+0100 gunicorn[844]:     self.lgpio.gpio_claim_output(self._handle(), pin_num, level=level)
>     2026-11-03T21:45:03+0100 gunicorn[844]:   File "/usr/lib/python3/dist-packages/lgpio.py", line 781, in gpio_claim_output
>     2026-11-03T21:45:03+0100 gunicorn[844]:     return _u2i(_lgpio._gpio_claim_output(handle&0xffff, lFlags, gpio, level))
>     2026-11-03T21:45:03+0100 gunicorn[844]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>     2026-11-03T21:45:03+0100 gunicorn[844]:   File "/usr/lib/python3/dist-packages/lgpio.py", line 458, in _u2i
>     2026-11-03T21:45:03+0100 gunicorn[844]:     raise error(error_text(v))
>     2026-11-03T21:45:03+0100 gunicorn[844]: lgpio.error: 'unexpected error'
>     2026-11-03T21:45:03+0100 gunicorn[844]: 2026-11-03 21:45:03,527 INFO app /home/pi/allsky/server/app.py:162: HTTP POST /gpio/digital -> 500
>     ```
>     
>     
>         
>           
>         
>     
>           
>         
>     
>         
>       
>     Hope this Helps

Thanks for that

Can you remove the overlay line

```
dtoverlay=pwm-2chan,pin=18,func=4,pin2=19,func2=4
```

Reboot and try it again

Thanks

Alex

drakula1981, 2026-11-04:
Fantastic! it works both with PWM mode enabled and disabled!
Perhaps you should add this in the module's information?

Alex-developer, 2026-11-04:
> 8\. Fantastic! it works both with PWM mode enabled and disabled!
>     Perhaps you should add this in the module's information?

Thats good, I'll get the documentation updated.

Earlier versions of Allsky relied on Hardware PWM, hence needing the overlay but I changed it to software PWM 

Alex

drakula1981, 2026-11-04:
Hardware PWM never worked for me, so many thanks for the very good jod!

EricClaeys, 2026-11-04:
@drakula1981,

Would you please upload `~/allsky/config/logs/support.log` ?

drakula1981, 2026-11-04:
Of course, here it is!
[support.log](https://github.com/user-attachments/files/34396271/support.log)

EricClaeys, 2026-11-05:
Thanks @drakula1981.  I created [Issue #5565](https://github.com/AllskyTeam/allsky/issues/5565) for `support.sh` creating a log file that's too big to upload.

Would you please run this and upload the results?
~~~
tree --du -h -ugp ~/allsky > /tmp/tree.txt
~~~

drakula1981, 2026-11-15:
@EricClaeys 
Sorry for the late answer.
Here it is :
[tree.txt](https://github.com/user-attachments/files/34872864/tree.txt)

EricClaeys, 2026-11-16:
@drakula1981,
Thanks.  Would you please do the following, if you still have one of the large support files:
~~~
cd ~/allsky/html/support
unzip -l NAME_OF_SUPPORT_FILE > unzip.txt    # Replace "NAME_OF_SUPPORT_FILE" with the support log file name
~~~
Then upload `unzip.txt`.  It will tell me the size of each file in the support log.